openhwfoundation/cvfpu #22: How to perform Div/Sqrt operation using the DIVSQRT unit

noureddine-as, 2020-02-13:
Hello,

I established a testbench that instanciates the fpnew_top, and I'm trying to understand how I can control the interface to feed the input operands and get the result back.

I use the following instructions inside an initial block. With MULADD operations I get the correct results, but when I change `fpu_op` to `fpnew_pkg::SQRT` or `fpnew_pkg::DIV` I get all zeroes at the output.

I use the same Implementation and Feature parameters used inside the EX stage of RI5CY. Simulation is performed using ModelSim 10.2c. _Do you have any idea why i don't get the results for Div and Sqrt? What of ModelSim does work with your SystemVerilog testbenches?_

```systemverilog
        // The input operands
        apu_operands_i[0] = 32'b00111111100000000000000000000000; // 1.0
        apu_operands_i[1] = 32'b01000000001000000000000000000000; // 2.5
        apu_operands_i[2] = 32'b00111111100000000000000000000000; // 1.0

        // Operation to be executed
        fpu_op        = fpnew_pkg::ADD;
        fpu_vec_op    = 1'b0;
        fpu_op_mod    = 1'b0; // fpu_op = ADD && fpu_op_mod = 0 ==> ADDITION 
                                           // fpu_op = ADD && fpu_op_mod = 1 ==> SUBTRACTION 

        // Input/Output formats
        fpu_int_fmt   = fpnew_pkg::INT32;
        fpu_src_fmt   = fpnew_pkg::FP32;
        fpu_dst_fmt   = fpnew_pkg::FP32; // By default, dest types = src type. fpnew_pkg::FP32;
        fp_rnd_mode   = fpnew_pkg::RNE;

        // Ready/Valid Interface
        in_valid_i = '1;
```

I have also got several warnings, that I just suppressed, _could these affect the results of the simulation?_ 

`vlog-2583: [SVCHK] - Some checking for conflicts with always_comb and always_latch variables not yet supported. Run vopt to provide additional design-level checks.`

Thank you very much in advance.
Best regards,

stmach, 2020-02-14:
Hello and thanks for your interest in this project!

From the get-go I can think of two potential problems:
1. In RISCY there is an option to turn off the DIV/SQRT unit through the parameters, so it might be you copied from a version with disabled DIV/SQRT. Would you mind pasting your version of the `Implementation` paramter you're using? Alternatively (for starting) you can also use the default implementation parameter from the package called `fpnew_pkg::DEFAULT_NOREGS` which enables all units and doesn't add any pipeline stages.
2. The DIV/SQRT hardware unit itself is contained in another repository, [pulp-platform/fpu_div_sqrt_mvp](https://github.com/pulp-platform/fpu_div_sqrt_mvp) which has to be compiled as well. Since you don't seem to have any simulation issues though I assume you compiled it as well.

The warning you see should not be a problem. It pertains to I believe arrays (I don't remember precisely) having some indexes assigned inside of an `always_comb` block and some outside of it, which the simulator cannot statically verify is safe before doing elaboration (since the dimensions are parametric)

Cheers

noureddine-as, 2020-02-14:
Thank you for your valuable support!

- Here is how I proceeded: I used bender to generate the synthesis script for ModelSim, a priori no dependencies problems and everything compiles fine, including the modules from `fpu_div_sqrt_mvp`.
- I have written a testbench where I instanciate an `fpnew_top` module, with the following parameters I extracted from the RI5CY core. There was a `C_DIV`parameter which activates `fpnew_pkg::MERGED` unit type in case it is enabled, so I just put that directly. So ADD/MUL operations are working, but not the DIV/SQRT.
- I tried with the `fpnew_pkg::DEFAULT_NOREGS` implementation, as well as with the standard  `fpnew_pkg::RV32F` feature, still not output.
- I also checked the `NUM_OPGROUPS` parameter in `fpnew_pkg`, which determines what blocks should be generated. It is set to the default `NUM_OPGROUPS = 4` value.
- I wasn't able to print the schematic in ModelSim to check if the DIV/SQRT block is taken into consideration or not. Someone suggested me that it is possible that the version of modelsim I have (SE, 10.2c) doesn't support SystemVerilog, do you recommend a particular ModelSim version?

```systemverilog

// -----------
// FPU Config
// -----------
// Features (enabled formats, vectors etc.)
parameter fpnew_pkg::fpu_features_t FPU_FEATURES_RI5CY = '{
        Width:         C_FLEN,
        EnableVectors: C_XFVEC,
        EnableNanBox:  1'b0,
        FpFmtMask:     {C_RVF, C_RVD, C_XF16, C_XF8, C_XF16ALT},
        IntFmtMask:    {C_XFVEC && C_XF8, C_XFVEC && (C_XF16 || C_XF16ALT), 1'b1, 1'b0}
        };

// Implementation (number of registers etc)
parameter fpnew_pkg::fpu_implementation_t FPU_IMPLEMENTATION_RI5CY = '{
        PipeRegs:  '{// FP32, FP64, FP16, FP8, FP16alt
                        '{C_LAT_FP32, C_LAT_FP64, C_LAT_FP16, C_LAT_FP8, C_LAT_FP16ALT}, // ADDMUL
                        '{default: C_LAT_DIVSQRT}, // DIVSQRT
                        '{default: C_LAT_NONCOMP}, // NONCOMP
                        '{default: C_LAT_CONV}},   // CONV
        UnitTypes: '{'{default: fpnew_pkg::MERGED}, // ADDMUL
                        '{default: fpnew_pkg::MERGED},  // DIVSQRT
                        '{default: fpnew_pkg::PARALLEL}, // NONCOMP
                        '{default: fpnew_pkg::MERGED}},  // CONV
        PipeConfig: fpnew_pkg::AFTER
        };
```

stmach, 2020-02-14:
Thanks for the detailed description of your setup, there doesn't seem to be any issue with that.

Just to make sure, as you're saying the unit outputs only `0`s for `DIV` operations. Does the unit complete a valid transaction on the output interface (i.e. `out_valid_o` goes high) with `'0` at the outputs? I'm just asking because the DIV operations are computed on an iterative unit that takes quite some cycles to produce a result, unlike the MULADD units that are fully pipelined. Before that (until there valid output is signalled by `out_valid_o = 1'b1`), the result might show all zeroes.

Concerning the Modelsim version, I currently can't check with an older version, but I know it to work for sure on 10.7b and it should also work on some previous versions. However you never know with these tools, I'll have to check once the older versions run again (likely next week).

Cheers

noureddine-as, 2020-02-17:
Hello again,
I did further tests, and here is my state of progress:
> Just to make sure, as you're saying the unit outputs only `0`s for `DIV` operations. Does the unit complete a valid transaction on the output interface (i.e. `out_valid_o` goes high) with `'0` at the outputs?

Actually when I perform ADDMUL operations the valid signal goes high.
[image]

However, when I change to SQRT (or DIV), the transaction does not complete
[image]

I think since these are iterative algorithms, they should provide 23-bit mantissa results in 23-ish number of cycles normally.

> I'm just asking because the DIV operations are computed on an iterative unit that takes quite some cycles to produce a result, unlike the MULADD units that are fully pipelined. Before that (until there valid output is signalled by `out_valid_o = 1'b1`), the result might show all zeroes.

I suspected that at the beginning, but even when I wait for 600 i get nothing, knowing that the `out_ready_i` signal is forced to `1` (just as in RI5CY). 
```systemverilog
      for(int unsigned i=0; i< 600; i++)
      begin
            @(posedge  clk);
            $display("Iteration %d",i);
      end
```

> Concerning the Modelsim version, I currently can't check with an older version, but I know it to work for sure on 10.7b and it should also work on some previous versions. However you never know with these tools, I'll have to check once the older versions run again (likely next week).

Actually I had to perform a few changes on the source code to get it work on my 10.2c ModelSim version. For example, I had to change the `localparam` key word in modules declarations to `paramater` and I had also some issue such as: 
```systemverilog
    assign index_lut[j] = NUM_LEVELS'(unsigned'(j));  // Original line in common_cells/src/lzc.sv
    assign index_lut[j] = (NUM_LEVELS)'(unsigned'(j)); // To overcome ModelSim errors, i put the parameter between parentheses.
```

For a newer version (10.5c - 2016) of ModelSim, I didn't have to do these changes in source code for it to compile, but the testbench was no longer working, even with ADDMUL. In this case I get the following, where nor the output nor the ready signal are defined.

[image]

Thanks again for your support .
Best regards

stmach, 2020-02-18:
Hello,

The reset signal `rst_n` is active-low (denoted by suffix `_n`). You need to tie it high in order for the sequential parts of the circuit to operate properly. Let me know if inverting your reset fixes the issue.

Cheers

stmach, 2020-02-18:
> and I had also some issue such as:
> 
> ```systemverilog
>     assign index_lut[j] = NUM_LEVELS'(unsigned'(j));  // Original line in common_cells/src/lzc.sv
>     assign index_lut[j] = (NUM_LEVELS)'(unsigned'(j)); // To overcome ModelSim errors, i put the parameter between parentheses.
> ```

Thanks for pointing this out, some tools have their issues when params used for bit-width aren't enclosed in parentheses (even though it's legal SystemVerilog AFAIK). Will fix it inside `common_cells`.

noureddine-as, 2020-02-18:
Hello,

Oh my bad, please pardon my rookie mistakes. It's working now. I get the result at the 10th cycle for Sqrt  (is it normal ?). This is using ModelSim 10.2c (2013).

So I think think this issue is resolved. Thanks a lot for the help!

[image]

However, using ModelSim 10.5c (2016) the simulation stops at a property labeled `gnt_idx` that wasn't verified. I report the result of the simulation here for reference. This error occured with both source code versions i.e. the original source code and the one where I made the changes mentioned earlier (putting parentheses around bit-widths + changing localparam to parameter)
```modelsim
# Iteration         10
# Iteration         11
# ** Fatal: Idx_o / gnt_o do not match.
#    Time: 270 ns Started: 270 ns  Scope: fpnew_tb.i_fpnew_bulk.i_arbiter.genblk1.gnt_idx File: ../../fpnew/src/common_cells/src/rr_arb_tree.sv Line: 231
# ** Note: $finish    : ../../fpnew/src/common_cells/src/rr_arb_tree.sv(231)
#    Time: 270 ns  Iteration: 6  Region: /fpnew_tb/i_fpnew_bulk/i_arbiter/genblk1/gnt_idx
# 1
# Break in VlGenerateBlock genblk1 at ../../fpnew/src/common_cells/src/rr_arb_tree.sv line 231
```
Here is the trace I got when the properties in the bottom of the `rr_arb_tree.sv`file are deactivated:
[image]

Thank you very much for your help.
Best regards

stmach, 2020-02-18:
Oof, indeed the `valid_o` signal should not be `x`, and it's strange that this behavior is Modelsim version-dependent.
I'll investigate this and if needed, will open an issue in the `common_cells` repo. Leaving this issue open for now.

noureddine-as, 2020-02-21:
Hello again,
I'm sorry for bothering, but I guess this question is related to this issue.
After having the RTL simulation worked, I tried a post-synthesis one using the same testbench and the same ModelSim 10.2c 2013 version. Synthesis is done using Design Compiler (2017 version) with a 350nm library.

Design Compiler produced a Verilog file output plus a SystemVerilog wrapper. These two with my testbench compiles well. So using the same testbench (with the active-low reset `rst_n`), I get all `X` in the outputs (`out_valid_o`, `in_ready_o`, `apu_result`, and `fpu_ff_flags_o`) as follows:
[image]

But when I invert `rst_n` at the input of the instaciated module, I get the `out_valid_o` to be zero (so at least it's defined) but never goes high and I never get the output result.
[image]

So I thought there was something I missed with the synthesis. I specified a clock of 350Mhz, without input nor output delays. From the reports I can see that the constraints are met. Should I somehow specify something to Design Compiler about the reset stuff? Or it's normally supposed to understand that? Any ideas?

Sorry again for my beginner questions.
Thank you in advance.
Best regards.

stmach, 2020-02-22:
In the first screenshot, it seems you're sampling all `X` inputs into the unit on the first active clock after reset, marked in yellow here:
[image]
Please make sure all input signals are defined before reset ends and let me know if that fixed your issue. _(some extended blurb about this at the end of the reply)_

On the different synthesis results, the snitch config includes 1 pipeline register for all units, so probably that's one place where they might differ.

Cheers

---
These `X` values sampled in will poison all downstream logic as any logic operation with `X` can yield `X` again, especially if the simulator is pessimistic about the handling of `X`es. Also, (System)Verilog is inherently bad for modelling uninitialized vs. undefined behavior as it knows only 4 logic values - but since we're sampling X into a register that probably doesn't have much to do with this issue.
The inverted reset (which keeps the circuit in reset at all times) actually brings the (resettable) registers in the circuit into a defined state again, so the control signal outputs at least show a defined value (since all control path registers are resettable).

noureddine-as, 2020-02-23:
You're right @stmach , defining the inputs before the reset ends resolved the problem!
I didn't know that, thank you very much for the detailed explanation!

Best regards

closed 2020-03-06 by stmach