openhwfoundation/cvfpu #23: fcvt.s.d gives wrong answer

b1f6c1c4, 2020-04-07:
I tried to run `fcvt.s.d` to convert a small double (`2.386004908190509275e-40` which is `0x37b4c8f800000000`) into single, but it gives `inf` (`0x7f800000`) instead of `2.38600491e-40` (`0x0002991f`).

stmach, 2020-04-10:
Hello,

Have you checked what is effectively applied to the FPU?
Setting up the operation with
```
op_i       = fpnew_pkg::F2F
rnd_mode_i = fpnew_pkg::RNE
src_fmt_i  = fpnew_pkg::FP64
dst_fmt_i  = fpnew_pkg::FP32
```
and applying your given input yields `0xffffffff0002991f` as a result, which is what you'd expect.
The issue is likely in the integration of the FPU into your system (either configuration or the application of control signals) then. What system are you using the FPU with? 

Cheers!

b1f6c1c4, 2020-04-10:
I'm trying with Ariane. How do we debug into it?