openhwfoundation/cvfpu #24: fcvt.s.d gives wrong answer

b1f6c1c4, 2020-04-15:
I tried to run `fcvt.s.d` to convert a small double (`2.386004908190509275e-40` which is `0x37b4c8f800000000`) into single, but it gives `inf` (`0x7f800000`) instead of `2.38600491e-40` (`0x0002991f`).

stmach, 2020-04-15:
A post-synthesis netlist obtained from Synopsys DC yields the correct result (simulated with Modelsim since I don't have VCS set up), so the issue seems contained to VCS for now.

b1f6c1c4, 2020-05-05:
Any update on this?

b1f6c1c4, 2020-05-28:
@stmach @zarubaf Well I'm wondering how can I help for this issue. Will it be helpful if we could agree on a certain version (commit SHA-1) of Ariane and we run VCS and ModelSim separately, and then compare their results? Due to some limitation (broken ModelSim license) I cannot do this on my end. So if you think it will help, you can give me a SHA-1 and I'll do VCS on my end. Otherwise you can debug yourself.

commit 79d83599 (2020-06-02): :bug: Fix missing typecast for VCS sim in casts

A missing typecast to `signed` caused VCS to trip the overflow condition
instead of the underflow condition for F2F casts in `fpnew_cast_multi`.

Fixes #24
closed 2020-06-02 by stmach