openhwfoundation/cvfpu #47: Mismatching Calculations

M31581, 2021-05-25:
I have been using the Imperas RISV Tests included with riscvOVPsim for compliance testing and having some queries on the F ISA extension tests.

OVPSim : [https://github.com/riscv-ovpsim/imperas-riscv-tests](url)
F ISA test download page: [https://www.ovpworld.org/library/wikka.php?wakka=riscvOVPsimPlus](url)

These tests from Imperas are the updated RISCV Compliance tests and the fpnew is have some mismatches on the FDIV, FMUL and FSQRT tests.

Here are the mismatching calculations:
[FPNEW_MISMATCH_COMPLIANCE.xlsx](https://github.com/pulp-platform/fpnew/files/6539765/FPNEW_MISMATCH_COMPLIANCE.xlsx)

I have only seen F ISA extension tests pass with the models (e.g. C/BFM), so this suggests there is a mismatch between the RTL and models.

Can you comment on my findings?

mp-17, 2021-11-22:
Hello @M31581,

Thanks for reporting. There are some rounding problems with the divider/square root units when used in combination with RNE policy. Indeed, there is a plan to update the entire `divsqrt` unit. 
Have you also found mismatches with `fmul` operations? I don't find them in the `.xlsl` file.

Best regards,
Matteo

M31581, 2021-11-23:
Hi @mp-17 

Thanks for confirming, good to here there is a plan to update the entire divsqrt unit.
Yes I have found mismatches in the fmul operations, the "I-FDIV-S-DYN-RDN-01 and I-FDIV-S-RDN-01" table is actually a typo it is the "I-FMUL-S-DYN-RDN-01 and I-FMUL-S-RDN-01".

I have just updated the table with correct info.

Regards,
Tommy