openhwfoundation/cvfpu #65: Benchmarking of PULP FPU against Whisper ISS

HamzaShabbir517, 2022-09-02:
To whom it may concern,

We are from Microelectronics Research Lab (MERL), based in Pakistan working on developing RISC-V based ASICs and SoCs.

Currently we were working on developing an Indigenously design RISC-V based Floating Point Unit (AI-FPU). Initially we were using PULP FPU as a benchmark for the verification of our designed AI-FPU. 

However, during the verification process we encountered few cases where the PULP FPU was not 100% compatible with Whisper ISS. A glimpse of those errors is shown in the figure attach.
[image]
[image]
[image]

Thank you,

Regards:
Engr Hamza Shabbir
Research Associate 
MERL

davideschiavone, 2022-09-02:
That's amazing, thanks @HamzaShabbir517 

The OpenHW Group team is already working in fixing the divider, but this document will definitively be very useful for them

@lucabertaccini @MikeOpenHWGroup @pascalgouedo @JeanRochCoulon @jquevremont

MikeOpenHWGroup, 2022-09-02:
Thanks for this @HamzaShabbir517!  As @davideschiavone indicated, the OpenHW Group has started a complete verification cycle of the CV32E40Pv2+CVFPU.  This effort is being led by @pascalgouedo.  In addition @lucabertaccini will be updating the DIV logic in the CVFPU.

This work has just begun, and we have encountered multiple issues (which will be captured as issues in this repository very soon).  Your input will be very useful to this effort.   As we get closer to completion of this verification effort, may we ask you to re-run your tests again?

HamzaShabbir517, 2022-09-02:
@davideschiavone thanks for the quick response. 
@MikeOpenHWGroup Sure please let me know as we have developed a Verification Environment in which we can generate Random Floating Point Numbers and we can also give millions of inputs to the design and compare the result against the ISS. 
A glimpse is shown below.

[image]

Regards:
Hamza Shabbir 
Research Associate 
MERL

stmach, 2022-09-19:
Hi there, 

Since `FNMADD` and `FNMSUB` both produce false results with an inverted sign bit in your screenshots, can it be that the two ops are swapped in their encoding?

Cheers

HamzaShabbir517, 2022-09-19:
@stmach Yes that's the issue the opcodes have been swapped