openhwfoundation/cvfpu #69: Narrowing converting instruction mismatch issue for number infinity

Shafi10x, 2023-01-23:
If a number is infinity in **FP64**, after converting it into **FP32**, will it be _infinity_ or of _max magnitude_? According to spike, it is infinity but according to RTL implementation, it is of max magnitude._Overflow_ and _infinity_ are handled together i.e saturating it or making it max magnitude. I think it is fine to handle overflow like this, but for infinity, there should be another case i.e _infinity should remain infinity._

pascalgouedo, 2023-01-23:
In principle I would agree that infinity stays infinity whatever operation happens to it...

stmach, 2023-01-23:
Indeed, `inf` should always map to `inf`, no matter the rounding mode or format in use. Under which circumstances do you observe max magnitude being returned? A quick glance into the RTL indicates that this issue should only happen when casting `inf`s with round mode `RDN` or `RTZ` (which should be fixed, of course).

Shafi10x, 2023-01-24:
In my case it was `RTZ` mode. I think the following part could be fixed. 
```
      // Overflow or infinities (for proper rounding)
      if ((destination_exp_q >= signed'(2**fpnew_pkg::exp_bits(dst_fmt_q2))-1) ||
          (~src_is_int_q && info_q.is_inf)) begin
        final_exp       = unsigned'(2**fpnew_pkg::exp_bits(dst_fmt_q2)-2); // largest normal value
        preshift_mant   = '1;                           // largest normal value and RS bits set
        of_before_round = 1'b1;
```