openhwfoundation/cvfpu #72: Bug report: wrong sign in fmul.d with RDN rounding mode

flaviens, 2023-02-15:
Hi there!

I found a bug in FPnew (bug found through Ariane).

The floating multiplication gives wrong results for some inputs with the RDN rounding mode.

For example, the  pseudo assembly code below executes incorrectly:
```
ft0 = 0x0 # corresponds to value +0.0
ft1 = 0x00000000dc98da8a # corresponds to value +1.82854e-314

fmul.d ft2, ft1, ft0
```

In the end of the snippet:  
- We expect the result to be 0x0, i.e., +0.0
- We get the result          0x8000000000000000, i.e., -0.0

pascalgouedo, 2023-02-16:
Found by formal verification as well [cv32e40p #728](https://github.com/openhwgroup/cv32e40p/issues/728)

flaviens, 2023-02-16:
Hi @pascalgouedo thanks! I add it to the list in the initial post.

closed 2023-03-02 by flaviens