openhwfoundation/cvfpu #83: Conversion from FP32 -2^31 to INT32 sets NV status

pashenden, 2023-03-31:
In v0.7.0, converting the FP32 value -2147483648.0 to INT32 produces the result 0x80000000 and sets the NV status flag. However, the input value can be exactly represented as an INT32.

Test case:

      operands_i = '{0: 32'hcf000000, 1: 32'h00000000, 2: 32'h00000000};
      rnd_mode_i = RNE;
      op_i = F2I;
      op_mod_i = 1'b0;
      src_fmt_i = FP32;
      int_fmt_i = INT32;

Root cause: 
[These lines](https://github.com/openhwgroup/cvfpu/blob/12c32ec71e3ee359a56f017eba139b6c281381dc/src/fpnew_cast_multi.sv#L443-L445) do not take account of the fact that for the valid input value, the exponent is exactly 31. If the sign is negative and the mantissa is 0x80000000 in this case, there should not be overflow.

pascalgouedo, 2023-03-31:
Hi @pashenden

It seems same kind of bug as already reported [here](https://github.com/openhwgroup/cv32e40p/issues/727).
But maybe not the same source line to correct.

Thanks for reporting.

pashenden, 2023-04-02:
Yes, same bug. In that issue, the operand value is 0xcf000000, and the fcvt.w.s instruction would invoke the F2I operation in the FPU. So fixing the FPU should address the bug in the processor core.

gregdavill, 2023-08-03:
@pascalgouedo Can this issue be closed now with #97 merged?

pascalgouedo, 2023-08-04:
Hi @gregdavill 
The work in on-going on bug resolution w/ formal verification guy.
We will close all resolved opened issue once definitely confirmed resolved by formal runs launched by him.

zarubaf, 2023-10-26:
Fixed with release v0.8.1.

closed 2023-10-26 by zarubaf