openhwfoundation/cvfpu #86: Bug report: multifmt slice uses wrong FP width for third operand

michael-platzer, 2023-04-18:
Hi,

the `fpnew_opgroup_multifmt_slice` uses the wrong FP width for the third operand.

AFAIK the third operand is only used by the `ADDMUL` opgroup, which expects that `operands_i[0]` and `operands_i[1]` have the FP format specified by `src_fmt_i`  and `operands_i[2]` and the result have the FP format `dst_fmt_i`:

https://github.com/openhwgroup/cvfpu/blob/5da3b411f3309d41b39dbc68da7449456f6be444/src/fpnew_fma_multi.sv#L36-L37

Accordingly, the `fpnew_opgroup_multifmt_slice` should use the width of FP format `dst_fmt_i` when assigning the individual elements of `operands_i[2]` to the lane instances. However, the width of `src_fmt_i` is used for all operands:

https://github.com/openhwgroup/cvfpu/blob/5da3b411f3309d41b39dbc68da7449456f6be444/src/fpnew_opgroup_multifmt_slice.sv#L181-L183

As a result, the third operand is incorrect for all lanes (except for the first lane), as shown in the following waveform. The first group of signals shows the `src_fmt_i` (which is `FP16`) and `dst_fmt_i` (which is `FP32`), as well as `operands_i[0]` (for reference) and `operands_i[2]` of the `fpnew_opgroup_multifmt_slice`. The elements of `operands_i[0]` which are 16 bits wide are correctly assigned to the individual `fpnew_fma_multi` instances (as seen in the signal groups below, which show the input operands of the first three instances). However, the elements of `operands_i[2]` which are 32 bits wide are not correctly assigned. The individual elements of these operands are surrounded by rectangles of different colors.

[image]

michael-platzer, 2023-04-18:
Testbench to reproduce the waveforms above: https://gist.github.com/michael-platzer/67c1c04eb0d699f2526166881867c250

MikeOpenHWGroup, 2023-04-18:
Hi @YoannPruvost, can we reproduce this issue in core-v-verif?

pascalgouedo, 2023-04-18:
Hi @MikeOpenHWGroup 

We are not using FP16 format in CV32E40P, only FP32.
So as src and dest format are both 32-b, they are interchangeable without being noticed by any tools/simulation test and even formal...

Cheers,
Pascal.

PR #87 (2023-04-19): Fix width of third local operand of multifmt slice
Hi,

this is a proposal to fix #86 by using the width of FP format `dst_fmt_i` for assigning the third local operand in the `fpnew_opgroup_multifmt_slice`.

With this patch the operands of the individual `fpnew_fma_multi` instances are assigned correctly, as shown in the waveform below.

[image]
closed 2023-04-19 by lucabertaccini