openhwfoundation/cvfpu #105: how complicated would it be add FP8 E3M4 and E2M5 support to this IP?

muzafferkal, 2023-11-02:
Hi,
Currently the only FP8 support is E5M2. How complicated would it be to add E3M4 and E2M5 support? Can I make a contribution to some organization to get these added? What would it take :-) ?

zarubaf, 2023-11-06:
Afaik, making this FPU extendable was one of its primary purposes. You should just need to add your formats to the package [here](https://github.com/openhwgroup/cvfpu/blob/d6e581628f3517a1fb1257507d3214e599f7859d/src/fpnew_pkg.sv#L36-L57). You can designate the amount of exponent and mantissa bits per format. Note that I am not the author though :-) Let me know whether this answers your question and we can close the issue.

fayalalebrun, 2024-03-15:
What about e5m2_fnuz? Is there a way to disable infinities and use a special NaN encoding?