openhwfoundation/cvfpu #107: DC Synthesis for cvfpu (fpnew) occurs fatal error

qleenju, 2024-03-05:
When I tried to synthesis FPnew using Design Compiler (the top module is `fpnew_top` or `fpnew_opgroup_block`),  the following error will always appear: "The tool has just encountered a fatal error", "Fatal: Internal system error, cannot recover". 
[image]

I looked for some corresponding solutions, but they didn't help. (like: https://www.edaboard.com/threads/fatal-internal-system-error-cannot-recover.203320/)

Does anyone know the reason? For example, is the DC version I am using too low?

MikeOpenHWGroup, 2024-03-05:
> is the DC version I am using too low?

Yes, 2014 was a long time ago.  Please try a more up-to-date version.

qleenju, 2024-03-05:
> > is the DC version I am using too low?
> 
> Yes, 2014 was a long time ago. Please try a more up-to-date version.

Thank you for your kindly reply. Currently I have no access to a higher version of Design Compiler, so I cannot verify it yet.

But today I spent lots of time trying to figure it out, and then I found some potential problems, that is, our lower version of DC may not support part of the SystemVerilog syntax.

For example, the output of function `any_anabled_multi` in `fpnew_pkg.sv` cannot be assigned to localparam logic values (the function is used in `fpnew_opgroup_block` module).  However, after I modified it as follows, there were no errors when assigning the output to localparm logic values.
[image]

I found the following functions are also not supported well by my low-version DC.
[image]

I'm not sure whether my guess is correct. But for now, since I can't use a higher version of DC, I may only try to modify the implementation of these functions.

MikeOpenHWGroup, 2024-03-05:
At the OpenHW Group our goal is to create "industrial grade" IP, so we will tend to use the most advanced code constructs that are supported by industrial tools.  We are not in a position to help you directly with this issue, but it seems you are on the right track.

I cannot say whether this will help, but there is an open-source SystemVerilog to Verilog translator called [sv2v](https://github.com/zachjs/sv2v) that might be useful to you (I have never used it).

Unlesss you have anything else, please close this issue.

qleenju, 2024-03-06:
Thank you so much. I'll try using that tool or something else, to resolve the imcompatibility issue with lower version DC.

closed 2024-03-06 by qleenju