openhwfoundation/cvfpu #110: [BUG] Incorrect Result from 'fdiv' Operation Leading to Negative Infinity Instead of Smallest Negative Number

youzi27, 2024-04-22:
### Bug Description

When performing floating-point division using the `fdiv.d` instruction in CVA6, under specific conditions where the expected result is the smallest representable negative double-precision floating-point number (`0xffefffffffffffff`), CVA6 incorrectly returns negative infinity (`0xfff0000000000000`). 

**Steps to Reproduce:**
1. Load `fa7` with `0xc024000000000000`.
2. Load `fs3` with `0x00000000000002cc`.
3. Execute the instruction: `fdiv.d ft6, fa7, fs3`.
4. Observe the resulting value in `ft6`.

**The log is as follows::**
```
core   0: 0x00000000800052b2 (0x233987d3) fsgnj.d fa5, fs3, fs3
1 0x00000000800052b2 (0x233987d3) f15 0x00000000000002cc
core   0: 0x00000000800052b6 (0x23188853) fsgnj.d fa6, fa7, fa7
1 0x00000000800052b6 (0x23188853) f16 0xc024000000000000
core   0: 0x00000000800052ba (0x1b38f353) fdiv.d  ft6, fa7, fs3
1 0x00000000800052ba (0x1b38f353) f 6 0xfff0000000000000
```

See: https://github.com/openhwgroup/cva6/issues/2060

pascalgouedo, 2024-09-06:
Hi,
There are no information about the cvfpu version or hash number and nothing about cvfpu configuration used when finding this bug so it is quite hard to answer.
But looking at some comments it seems original PULP DIV/SQRT unit is used.
This unit had some bugs reported when using Single-Precision format so another DIV/SQRT unit from T-Head E906 core was added to replace it but it was supporting SP only (v0.8.1 tag).
As this PULP DIV/SQRT unit had some bugs when using Double-Precision format as well, another unit from T-Head C910 was added to cope with multi-format support (FP64, FP32, FP16 and SIMD operations).
This is available in develop branch.
To use it top level parameter should be assigned like that :

```
module fpnew_top #(
  // FPU configuration
  // DivSqrtSel chooses among PULP, TH32, or THMULTI (see documentation and fpnew_pkg.sv for further details)
  parameter fpnew_pkg::divsqrt_unit_t       DivSqrtSel     = fpnew_pkg::THMULTI,
```

youzi27, 2024-09-06:
Please see the answer in #111.