openhwfoundation/cvfpu #118: [BUG] Unexpected Result When fsqrt.s Processes Negative Floating-Point Numbers

fly-1011, 2024-05-13:
**Bug Description**
When the `fsqrt.s` instruction is executed on a negative double-precision floating-point number, the resulting output differs from that of Spike. Spike returns `0x7fc00000`, which represents a NaN value.

How to reproduce and the logs are as follows:
Example:
Initialize `ft0 `with `0xC08140003F800000` (a negative double-precision floating-point value) and execute `fsqrt.s ft2, ft0` .

Spike results are: `ft2= 0x7fc00000`
CVA6 results are: `ft2= 0x3f800000`

See:https://github.com/openhwgroup/cva6/issues/2100

MikeOpenHWGroup, 2024-05-13:
Thanks for your Issue @fly-1011.  Just FYI, the CVA6 is using a rather out-dated version of CVFPU and is not actively verifying the floating point ISA beyond running the Architectural Compliance Tests.  The first full release of the CVA6 (known as the [CV32A65X](https://docs.openhwgroup.org/projects/cva6-user-manual/04_cv32a65x_design/source/index.html)) does not support floating point.

@pascalgouedo, does this issue impact the CV32E40P v2.0.0?

pascalgouedo, 2024-05-13:
Hi,
What looks strange to me in the bug description is that is it mentioning fqrst.s (single precision computation using single-precision values) together with negative double-precision floating-point number.
As explained by RISC-V spec, when D extension is used, f registers are widened to 64 bits and can hold either 32-bit or 64-bit floating point values.
When f registers are used with single-precision instructions (\*.s), only 32 LSBs are used.
With double-precision instructions (\*.d), all 64 bits are used.

So CVA6 and CVFPU behavior described in [cva4#2102](https://github.com/openhwgroup/cva6/issues/2102) looks good to me.

And by the way CV32E40P is not supporting D extension.

fly-1011, 2024-05-14:
> Hi, What looks strange to me in the bug description is that is it mentioning fqrst.s (single precision computation using single-precision values) together with negative double-precision [...]

Thank you for your response. I haven't found explicit documentation in the RISC-V specifications regarding single-precision instructions using only the lower 32 bits when handling double-precision floating-point numbers. It appears that in its operations, the CVA6 only utilizes the lower 32 bits. I will be closing this issue.

closed 2024-05-14 by fly-1011