openhwfoundation/cvfpu #119: [BUG] fdiv incorrectly returns infinity when dividing by NaN

fly-1011, 2024-05-13:
**Bug Description**
When executing the `fdiv.s ft1, ft4, ft6` instruction, if the initial value of `ft6` is `0x7ff8000000000000 (NaN)`, the result in `ft1` on CVA6 is `0x7f800000 (infinity)`. However, in Spike, the result for `ft1` is `0x7fc00000 (NaN)`.

**Steps to Reproduce:**

Initialize `ft4` with `0x3fe000005f000000`
Initialize` ft6` with `0x7ff8000000000000(nan)`
Execute the instruction: `fdiv.s ft1, ft4, ft6.`
Observe the resulting value in `ft1`.

**The log is as follows:**

```
core   0: 0x0000000080002018 (0x186270d3) fdiv.s  ft1, ft4, ft6
3 0x0000000080002018 (0x186270d3) f 1 0xffffffff7f800000
```
It appears that the CVA6 uses only the **lower 32 bits** of `ft4` and `ft6` for computations, resulting in the output `0x7f800000 (infinity)`. The implementation does not correctly handle double-precision NaN inputs; it does not properly identify `ft6` as `NaN`, instead erroneously truncating the data.

See:https://github.com/openhwgroup/cva6/issues/2102

MikeOpenHWGroup, 2024-05-13:
See my comments made for #118.

pascalgouedo, 2024-05-13:
See my comment in #118

Not an issue, just read the doc...